QIDITECH/QIDIStudio #116: Extruder stops extruding (2.02.00.60)

jcdevzero, 2025-08-16:
Upgraded to v2.02.00.60 today and was attempting to print the **LPT Spool** part of the **3D Printable Jet Engine** here:

https://www.thingiverse.com/thing:1327093/

Using QIDI rapido PLA on a Plus 4. Default built in settings for this filament type. The part was oriented large end on the bottom and using tree supports.

First half of the print was perfect. After the top of the tree supports, one additional interface layer is printed, and then all of a sudden the extruder stops extruding any material even though the print head continues to move. It then continues the motions for each layer but no material is extruded.

It is not a clog as I canceled the print, cleaned the plate, and immediately tried again. Again, the first half of the print was perfect and then after the tree supports and the interface layer, again it stopped extruding while the print head continues to move and the printing process continues.

I deleted the model and re-imported it, sliced it, and tried again. Again, same failure in the same spot, ie. after the tree supports and the interface layer.

If I look at the **Preview** of the layers, everything looks good. All the layers are there. But for whatever reason, the extruder stops extruding once it has printed the interface layer above the tree supports. I failed 4 times and I gave up.

At this point I downgraded to the previous version v2.01.01.52 and tried again. With this version the part prints perfectly.

I think there might be some bug in v2.02.00.60?

wjyLearn, 2025-08-16:
Can you provide the GCode obtained from two different versions of the slices? So that we can look for the problem.

jcdevzero, 2025-08-20:
After some more playing with it, I don't think it's the release as I just reproduced with an older release version. My current guess is that the extruder is overheating. I will close this issue for now and reopen if I get more data. Thanks.

closed 2025-08-20 by jcdevzero