QIDITECH/QIDIStudio #118: Can`t send print job with BOX

Poffertjee, 2025-09-01:
When I click print plate and then send tsk to the printer I have error "Failed to set the box printing slot..." 
Printer QIDI Plus 4

[image]

wjyLearn, 2025-09-03:
This error is due to the failure to issue the instruction for the corresponding channel of the box to the printer. I think you need to check the connection status of the box.

wjyLearn, 2025-09-04:
If possible, you can upload the log of the software sending failure so that we can analyze the problem.

rustyg777-ship-it, 2025-10-01:
I don't HAVE my box yet, ( seems to be lost in shipping, haha ) but I am getting the same error message.  But now I can't print.  I have clicked the sync button and tries to sync a box that is not connected because I don't have it.  Please fix this, I am having to copy files to a USB drive to print them now.

Poffertjee, 2025-10-03:
`2025-10-03 11:22:17.848457[Thread 0x00001e98]:Slic3r::GUI::Preview::load_print_as_fff 706: print is step done, posSlice 1, posSupportMaterial 1, psGCodeExport 1
2025-10-03 11:22:17.848457[Thread 0x00001e98]:Slic3r::GUI::GCodeViewer::load: the same id 68, return directly, result 0000025E73F91830 
2025-10-03 11:22:17.851453[Thread 0x00001e98]:Slic3r::GUI::MainFrame::get_enable_print_status: m_print_select 1, enable= 1 
2025-10-03 11:22:17.851453[Thread 0x00001e98]:Slic3r::GUI::MainFrame::get_enable_slice_status: m_slice_select 1, enable= 0 
2025-10-03 11:22:17.851453[Thread 0x00001e98]:Slic3r::GUI::MainFrame::update_slice_print_status m_slice_select 1: can_slice= 1, can_print 1, enable_slice 0, enable_print 1 
2025-10-03 11:22:17.851453[Thread 0x00001e98]:Slic3r::GUI::MainFrame::get_enable_print_status: m_print_select 1, enable= 1 
2025-10-03 11:22:17.851453[Thread 0x00001e98]:Slic3r::GUI::MainFrame::get_enable_slice_status: m_slice_select 1, enable= 0 
2025-10-03 11:22:17.851453[Thread 0x00001e98]:Slic3r::GUI::MainFrame::update_slice_print_status m_slice_select 1: can_slice= 1, can_print 1, enable_slice 0, enable_print 1 
2025-10-03 11:22:17.879035[Thread 0x00001e98]:OctoPrint: Error sending G-code: , HTTP 401, body: {"error": {"code": 401, "message": "Unauthorized", "traceback": "Traceback (most recent call last):\n\n  File \"/home/mks/moonraker-env/lib/python3.7/site-packages/tornado/web.py\", line 1692, in _execute\n    result = await result\n\n  File \"/home/mks/moonraker/moonraker/components/application.py\", line 505, in prepare\n    self.request, self.auth_required\n\n  File \"/home/mks/moonraker/moonraker/components/authorization.py\", line 918, in authenticate_request\n    raise HTTPError(401, \"Unauthorized, Force Logins Enabled\")\n\ntornado.web.HTTPError: HTTP 401: Unauthorized (Unauthorized, Force Logins Enabled)\n"}}`

i think the problem is http error 401 but i input ip adress and api key in printer tab

[image]

wjyLearn, 2025-10-27:
It has been optimized in the latest version.

closed 2025-10-27 by wjyLearn